toblum/McLighting #413: Nodemcu forgets settings

piscator83, 2019-06-07:
Not sure whether it's got something to do with ML or not, but whenever my Nodemcu is cut off from power it forgets wifi, pixel...settings. Seems like a kind of reset as I opens an AP and I have to configure everything again. Any ideas?

toblum, 2019-06-13:
@piscator83 Looke like McL can't store it's settings in the filesystem. Make sure you use the correct settings for "Flash Size" in the board settings (https://github.com/toblum/McLighting/wiki/Software-installation#setup-the-arduino-ide)

smoochiee, 2019-06-14:
i think its a bug...ive experienced it to...when that happens unplug and replug again...ive set spiffs to  4m (1m spiffs)

smoochiee, 2019-06-14:
problem is when ur hotspot is off and u plug esp..then turn on wifi it connects to mclightning ap..

smoochiee, 2019-06-14:
turn on hotspot first then plug esp...their will be no problem..

piscator83, 2019-06-25:
@smoochiee thx, that really helped. I would have never come to this conclusion by myself cause I didn't realize that my hotspot was down...(quite common when you move in ;-) )
But will it come back to the hotspot once that happened? Gotta try this.
If not, it feels a bit buggy...

toblum, 2019-06-26:
@smoochiee Good point!

@piscator83 Yes, it should come back after some timeout in the captive portal.

closed 2019-06-26 by toblum