toblum/McLighting #415: wrong WS2812FXStripSettings.stripSize

Talumux, 2019-06-15:
I am trying to use mclightning with a matrix of 48 by 27 LEDs (=1296 LEDs). But I got a wrong pixel count information when I run the code. The UART Output shows first the correct number (1296) but later on the last digit is cut off (129):

...
Reading neoconfig file... Opened!
Config:  Parsed!
{"pixel_count":1296,"rgb_order":82,"pin":14}Reading config file... Opened!
Config:  Parsed!
...
then later:
...
E1.31 mode setup complete.
Read cfg: {"mode":1,"strip_mode":0,"brightness":15,"speed":255,"red":255,"green":168,"blue":13} Success!
*WM: freeing allocated params!
Saving Strip cfg: {"pixel_count":129,"rgb_order":82,"pin":14}
...

I looked through the code an insert NUMLEDS as strip size in line 260 of McLighning.ino

`void initStrip(uint16_t stripSize = NUMLEDS, neoPixelType RGBOrder = WS2812FXStripSettings.RGBOrder, uint8_t pin = WS2812FXStripSettings.pin){`

instead of

`void initStrip(uint16_t stripSize = WS2812FXStripSettings.stripSiz, neoPixelType RGBOrder = WS2812FXStripSettings.RGBOrder, uint8_t pin = WS2812FXStripSettings.pin){`

as a quick fix.
It seems to work, but I think this is not the root cause of my problem.
Does anyone know how why the pixel count information is wrong in my case?

toblum, 2019-06-17:
@Talumux How did you set the number of LEDs? Using REST API? 
http://YOUR_MCL_IP/pixelconf?ct=1296

Talumux, 2019-06-24:
@toblum  I edited the definitions.h and set #define NUMLEDS 1296

toblum, 2019-06-26:
@Talumux Could you please try and set the number of LEDs using the call is posted above?

Talumux, 2019-06-26:
Hi, I set the number of LED using the pixelconf?ct=1296 command.
After that all LEDs are working as expected.  But after a reset (example power off), the wrong pixel size of 129 ist restored. The Output via Uart is (see the last line!):

`FS File: /neoconfig.json, size: 44B
FS Usage: 1004/957314 bytes

Reading neoconfig file... Opened!
Config:  Parsed!
{"pixel_count":1296,"rgb_order":82,"pin":14}Reading config file... Opened!
Config:  Parsed!
{"mqtt_host":"","mqtt_port":"","mqtt_user":"","mqtt_pass":""}WiFiManager config FS Read success!
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: Adding parameter
*WM: strip_size
*WM: Adding parameter
*WM: led_pin
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
Saving Strip cfg: {"pixel_count":1296,"rgb_order":82,"pin":14}
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.44
SaveConfig is False!
WiFiManager config FS Save failure!
connected...yeey :)
Open http://192.168.178.44/ to open McLighting.
Use http://LEDTable.local/ when you have Bonjour installed.
New users: Open http://192.168.178.44/upload to upload the webpages first.

E1.31 mode setup complete.
Couldn't find "/stripstate.json"
 Failure!
*WM: freeing allocated params!
SET_MODE: 0 0
Saving Strip cfg: {"pixel_count":129,"rgb_order":82,"pin":14}
`

toblum, 2019-06-29:
Hi @Talumux,

it seems like the config file couldn't be stored. What is returned when you call:
http://<YOUR_MCL_IP>/list?dir=/

Regards
Tobias

Talumux, 2019-06-30:
when I execute `http://<YOUR_MCL_IP>/list?dir=/ ` I receive
`[{"type":"file","name":"config.json"},{"type":"file","name":"stripstate.json"},{"type":"file","name":"neoconfig.json"}]`

toblum, 2019-07-01:
This looks OK. And what do you get when you call: http://<YOUR_MCL_IP>/neoconfig.json

Talumux, 2019-07-01:
Hi Tobias,
the output of http://<YOUR_MCL_IP>/neoconfig.json is various. After a reset the output is
`{"pixel_count":129,"rgb_order":82,"pin":14}`. After executing http://YOUR_MCL_IP/pixelconf?ct=1296 the output is `{"pixel_count":1296,"rgb_order":82,"pin":14}`
Thanks for your ideas so far!

Talumux, 2019-07-01:
Maybe you can have a look at the following output.
I set /pixelconf?ct=1296
I checked /list?dir=/
I checked /neoconfig.json
I did a hw reset (e.g. I opened the Arduino Serial monitor again) and get the following output

FS File: /style.css, size: 489B
FS File: /stripstate.json, size: 86B
FS File: /neoconfig.json, size: 44B
FS Usage: 2761/957314 bytes

Reading neoconfig file... Opened!
Config:  Parsed!
**{"pixel_count":1296,"rgb_order":82,"pin":14}Coudnt find config.json
WiFiManager config FS Read failure!**
*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: Adding parameter
*WM: strip_size
*WM: Adding parameter
*WM: led_pin
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
**Saving Strip cfg: {"pixel_count":1296,"rgb_order":82,"pin":14}**
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.178.28
**SaveConfig is False!
WiFiManager config FS Save failure!**
connected...yeey :)
Open http://192.168.178.28/ to open McLighting.
Use http://LEDTable.local/ when you have Bonjour installed.
New users: Open http://192.168.178.28/upload to upload the webpages first.

E1.31 mode setup complete.
Read cfg: {"mode":1,"strip_mode":1,"brightness":196,"speed":196,"red":67,"green":172,"blue":255} Success!
*WM: freeing allocated params!
Saving Strip cfg: {"pixel_count":**129**,"rgb_order":82,"pin":14}

toblum, 2019-07-05:
Hi @Talumux,

I think I found the cause of the issue. The setting was truncted to three numbers, so that 1296 resulted in 129 after restart. The issue should be fixed in the next release 2.2.5.

Many greeting
Tobias

toblum, 2019-07-05:
@Talumux I just published the new release: https://github.com/toblum/McLighting/releases/tag/v2.2.5

closed 2019-07-05 by toblum