toblum/McLighting #417: Hardware Schematic Question

VeniceNerd, 2019-06-18:
I got a question about your hardware schematic:

[image]

Why are you supplying power both via the VIN pin and the 3.3V pin? Wouldn't it be one or the other?

Thanks!

M4lik, 2019-06-19:
The 3v3 pin is not an "input" pin at; least not here. The 3V3 are needed for the level shifter as a reference voltage. Then you apply 5V to the other side of the level shifter as the second reference voltage.  The uppper side of this board is a "seperate" circuit.

closed 2019-06-19 by M4lik

VeniceNerd, 2019-06-19:
Interesting!!! But what are we level shifting for? I don’t see anything else connected that would make use of the 3.3V?

toblum, 2019-06-19:
The ESP is 3.3V, the strip works with 5V (although 3.3V works in most cases). This is why level shifting is a good idea. Also for protection.
Have a look here: https://learn.adafruit.com/adafruit-neopixel-uberguide/powering-neopixels#driving-5v-neopixels-from-3-dot-3v-microcontrollers-5-22