toblum/McLighting #419: Can't get my strip to light up :(

VeniceNerd, 2019-06-20:
Hey guys,

I finally got everything setup and was so excited to see my light strip light up, but nothing happened. :(

I successfully installed the software on the board and connected all the hardware:

[image]
[image]
[image]
[image]

This is the lightstrip I am using:
[https://www.amazon.com/gp/product/B079ZZ9BYH/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1](https://www.amazon.com/gp/product/B079ZZ9BYH/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1)

I can successfully connect to the Web Interface as well:
[image]

Here is my definitions.h

```
//#define USE_WS2812FX_DMA       // LED_PIN is ignored & set to RX/GPIO3  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART1     // LED_PIN is ignored & set to D4/GPIO2  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX
//#define USE_WS2812FX_UART2     // LED_PIN is ignored & set to TX/GPIO1  Uses WS2812FX, see: https://github.com/kitesurfer1404/WS2812FX

// Neopixel
#define LED_PIN 2       // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached
#define NUMLEDS 24       // Number of leds in the strip
//#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
#define BUILTIN_LED 2    // ESP-12F has the built in LED on GPIO2, see https://github.com/esp8266/Arduino/issues/2192
#define BUTTON 4         // Input pin (4 / D2) for switching the LED strip on / off, connect this PIN to ground to trigger button.

#define HOSTNAME "McLighting01"   // Friedly hostname

#define HTTP_OTA             // If defined, enable ESP8266HTTPUpdateServer OTA code.
//#define ENABLE_OTA         // If defined, enable Arduino OTA code.
#define ENABLE_AMQTT         // If defined, enable Async MQTT code, see: https://github.com/marvinroger/async-mqtt-client
//#define ENABLE_MQTT        // If defined, enable MQTT client code, see: https://github.com/toblum/McLighting/wiki/MQTT-API
#define ENABLE_HOMEASSISTANT // If defined, enable Homeassistant integration, ENABLE_MQTT or ENABLE_AMQTT must be active
#define ENABLE_BUTTON        // If defined, enable button handling code, see: https://github.com/toblum/McLighting/wiki/Button-control
//#define MQTT_HOME_ASSISTANT_SUPPORT // If defined, use AMQTT and select Tools -> IwIP Variant -> Higher Bandwidth
#define ENABLE_LEGACY_ANIMATIONS // Dont disbale this for now
#define ENABLE_E131              // E1.31 implementation
#define USE_HTML_MIN_GZ             //comment for using index.htm & edit.htm from SPIFFs instead of PROGMEM
//#define CUSTOM_WS2812FX_ANIMATIONS  //uncomment and put animations in "custom_ws2812fx_animations.h" 

#ifdef ENABLE_E131
  #define START_UNIVERSE 1                    // First DMX Universe to listen for
  #define END_UNIVERSE 2                      // Last Universe to listen for, starting at UNIVERSE
                                              // MUST: END_UNIVERSE >= START_UNIVERSE
#endif

//#define WIFIMGR_PORTAL_TIMEOUT 180
//#define WIFIMGR_SET_MANUAL_IP

#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif

#ifdef MQTT_HOME_ASSISTANT_SUPPORT
  #define MQTT_HOME_ASSISTANT_0_87_SUPPORT // Comment if using HA version < 0.87
#endif

#if defined(USE_WS2812FX_DMA) and defined(USE_WS2812FX_UART)
#error "Cant have both DMA and UART method."
#endif
#if defined(ENABLE_MQTT) and defined(ENABLE_AMQTT)
#error "Cant have both PubSubClient and AsyncMQTT enabled. Choose either one."
#endif
[... 105 lines cut ...]
  unsigned long time_statechange = 0;   // Time when the state last changed
  int timeout_statechange_save = 5000;  // Timeout in ms to wait before state is saved
  bool state_save_requested = false;    // State has to be saved after timeout
#endif
#ifdef ENABLE_STATE_SAVE_SPIFFS
  bool updateStateFS = false;
#endif

// Button handling
#ifdef ENABLE_BUTTON
  #define BTN_MODE_SHORT "STA| 1|  0|245|196|255|255|255"   // Static white
  #define BTN_MODE_MEDIUM "STA| 1| 48|245|196|255|102|  0"    // Fire flicker
  #define BTN_MODE_LONG "STA| 1| 46|253|196|255|102|  0"  // Fireworks random

  unsigned long keyPrevMillis = 0;
  const unsigned long keySampleIntervalMs = 25;
  byte longKeyPressCountMax = 80;       // 80 * 25 = 2000 ms
  byte mediumKeyPressCountMin = 20;     // 20 * 25 = 500 ms
  byte KeyPressCount = 0;
  byte prevKeyState = HIGH;             // button is active low
  boolean buttonState = false;
#endif
```

I would be so grateful if someone could take a look at this and see what I am doing wrong. When I connect power only the blue board light comes on. The strips don't do anything at all. I try to select different colors in the web interface and nothing happens either. So discouraged right now. Hope someone has any insight!

VeniceNerd, 2019-06-20:
I also just measured Voltage on various parts:

Bottom part of the board: 5.16V
Top part of the board: 3.29V
End of the LED strip (other side of where I am connecting my power and data): 5.05V

So it appears as if everything gets power?

VeniceNerd, 2019-06-20:
I uncommented this line:

`#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)
`
but it still doesn't seem to work.

VeniceNerd, 2019-06-20:
Alright, now I connected my data to PIN 1 on the board (as was shown on the hardware page):

[image]

And changed the LED pin line back to14:
`#define LED_PIN 14       // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached`

Now two random LED's on the strip actually light up:
[image]

But just those two and any settings on the web interface don't make any changes on the strip.

Wits end.

toblum, 2019-06-21:
Hi @VeniceNerd,

this should usually work as far as I can tell from the photos. 

The PIN and number of LEDs can now be configure via HTTP call: https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-pixelconf-http10

Please try http://<IP>/pixelconf?pin=14 and http://<IP>/pixelconf?ct=<number of LEDs>

If that diesn't work, please try out a simple strandtest script (e.g. https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino) to make sure the strip itself works and to check where the problem originates.

Regards
Tobias

jseadragon, 2019-06-26:
@VeniceNerd so, I'm going to give this a shot (I've literally been in your shoes). 

A few things just looking over your setup...
1. I'm not clear what you have connected on your data line. If is a fuse you're probably ok. If it's a resistor, check the resistance as I used a too high value and it caused some issues.
2. You mention that you're using `LED_PIN 14`, but your strip is connected to what appears to be `D1`. Is that correct?
```
#define LED_PIN 14 // LED_PIN (14 / D5) where neopixel / WS2811 strip is attached
```

If you're getting DC voltage across the board, then the most likely culprit is something on the data line (either noise or something else). 

When I built mine, I had failed in several ways before finally getting it right:
 * Wrong resistor value inline on the data line
 * Backwards connected data line (doh)
 * Unusual pins lighting up (actually it was quite similar to what you describe) when I was connected to the wrong data pin. 

I suggest re-flashing your ESP with the full code you have just to make sure you have your changes propagated. I made this mistake as well. I guess it's the side effect of working on this stuff late at night after a long day . :) 

I also suggest using a short strip with 4-5 LEDs on it for testing. I've been using that for validating my board wiring before doing an install somewhere in my house.

VeniceNerd, 2019-07-02:
> The PIN and number of LEDs can now be configure via HTTP call: https://github.com/toblum/McLighting/wiki/HTTP-REST-API#get-pixelconf-http10
> 
> Please try http:///pixelconf?pin=14 and http:///pixelconf?ct=

I'm so sorry, but I don't really know how to run those commands. Where do I need to go to run them?

> If that diesn't work, please try out a simple strandtest script (e.g. https://github.com/adafruit/Adafruit_NeoPixel/blob/master/examples/strandtest/strandtest.ino) to make sure the strip itself works and to check where the problem originates.

Sorry again, but what should I do with that script? Do I upload it somewhere?

VeniceNerd, 2019-07-03:
> @VeniceNerd so, I'm going to give this a shot (I've literally been in your shoes).

Thank you!!!!

> 1. I'm not clear what you have connected on your data line. If is a fuse you're probably ok. If it's a resistor, check the resistance as I used a too high value and it caused some issues.

The data line just goes from the Arduino to the strip. It's a very short cable so I currently do not have anything connected on the data line.

> 2. You mention that you're using `LED_PIN 14`, but your strip is connected to what appears to be `D1`. Is that correct?

That is correct. My strip is connected to D1. Should I be changing it to D1 in the code as well? I noticed that the tutorial on this site had D1 on the board but D14 in the software.

> * Backwards connected data line (doh)

What does that mean?

> I suggest re-flashing your ESP with the full code you have just to make sure you have your changes propagated. I made this mistake as well. I guess it's the side effect of working on this stuff late at night after a long day . :)

How do you re-flash the ESP with the full code? I have just been making changes in the definitions.h and then compile and re-upload.

Also: should I uncomment this line or leave it commented?

> //#define LED_TYPE_WS2811  // Uncomment if LED type uses 400 KHz (classic 'v1' (not v2) FLORA pixels, WS2811 drivers)

VeniceNerd, 2019-07-03:
Two more additions:

The strip definitely works. I connected it to one of those prebuilt WiFi controllers and all LED’s light up. So something is wrong in my setup. 

Maybe this video will show what I am doing wrong:
https://www.dropbox.com/s/i9zzdt664eaxkmx/Video%20Jul%2002%2C%2017%2047%2056.mov?dl=0

toblum, 2019-07-05:
Hi @VeniceNerd,

sorry for my late answer. Unfortunately there were a few instructions on the setup missing for the latest version. I added these to the wiki now: https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

So I think you just need to call:  
http://10.0.1.149/pixelconf?ct=300&rgbo=grb&pin=5
to tell McL that you're using 300 LEDs in GRB mode and that the daa pin is connected to pin 5 (D1).

Sorry for the confusion.

Best regards
Tobias

VeniceNerd, 2019-07-10:
> So I think you just need to call:
> http://10.0.1.149/pixelconf?ct=300&rgbo=grb&pin=5
> to tell McL that you're using 300 LEDs in GRB mode and that the daa pin is connected to pin 5 (D1).

YES! That finally lit my my LED strip!!! Out of curiosity, why is pin 5 on D1? Is there any rhyme or reason to it?

Also, I don't think the RGB mode is set correctly for my strip yet. Even when I set it to "static red" my strip looks like this:

[image]

My strip is actually an [RGBW RGB+Cool White SK6812 ](https://www.amazon.com/gp/product/B079ZZ9BYH/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)LED strip. 

Do I need to set a different mode for this strip? Also, how do you actually set the pure white channel in McLighting? I did install the developer branch which I believe had white mode support?

Thanks for all your help!

toblum, 2019-07-13:
Hi @VeniceNerd,

I'm not sure why they chose this pinout scheme, but you can use this as a reference: https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/

I think the RGB mode is OK as a wrong mode would only swap the colors (red would be green and so on). Does it react to any changes you do via UI? Seeing the serial output when you change the mode would be interesting.

I'm not sure with the white mode. I'm waiting for my new SK6812 strip to arrive from china. It might be that there is still some code missing to support RGBW as there is still a big PR https://github.com/toblum/McLighting/pull/364 pending. I'll have to figure that out too.

Regards
Tobias

VeniceNerd, 2019-07-14:
Hey Tobias,

I’d love to send you one of my SK6812 strips. Would be more than happy to contribute to this project. Especially since I’d really like support for these strips. ;)

Going by you name I assume you are in Germany? (I’m German too). If you can PM me your address I’ll send you a strip right over!

toblum, 2019-07-31:
@VeniceNerd Yes, I think that's the best option now if you want to use RGBW. Maybe we'll have a combined version someday, but for the short term, the mantioned version is great.

HiranChaudhuri, 2019-08-03:
> I think the RGB mode is OK as a wrong mode would only swap the colors (red would be green and so on). Does it react to any changes you do via UI? Seeing the serial output when you change the mode [...]

McLighting currently supports RGB strips only. This means it sends out 24 bits per pixel, which will be used as 8 bits for each of the three colours. SK6812 strips however have a fourth channel and thus use 32 bits per pixel. This is the reason why a configuration for solid red ends up in a coloured strip.

To make McLighting compatible with 32bits/pixel is actually documented here:
https://github.com/toblum/McLighting/pull/364

bpohvoodoo, 2019-08-08:
@toblum 
> Maybe we'll have a combined version someday, ...
It is already "combined"
See:
[image]

VeniceNerd, 2019-08-08:
@bpohvoodoo whaaaaat??? How do I get to that page???

bpohvoodoo, 2019-08-08:
If you use my version just click on the gear icon

deftdawg, 2019-08-13:
@VeniceNerd should probably close the issue, if you're up and running.

VeniceNerd, 2019-08-16:
I haven’t had time to try anything new yet. Will happen this weekend.

VeniceNerd, 2019-08-16:
@bpohvoodoo looks like the installation guide on your fork isn’t working. Links to a random Wordpress blog. Do I just follow the installation guide here but use your fork instead of the toblum version?

toblum, 2019-08-16:
@bpohvoodoo 
That looks great. I have tried it out recently and it looks great. Unfortunately I don't find the time at the moment to do something similar in my McLighting version. I have to pause the project a bit as I really don't find the time.
For the meanwhile I will post a note on the project readme and will also mention your fork for all people looking for more functionality. I hope this is OK for you.

Regards
Tobias

grumpazoid, 2020-03-31:
I just tried McLighting for the first time today.
I came across this post as my LEDs were not lighting.
My D1 mini clone had previously been running another well known firmware without issue.
The above solution of http://10.0.1.149/pixelconf?ct=300&rgbo=grb&pin=5 also got me up and running.

I am much enjoying the improved effects.

Question is why did it not take the correct pin from the sketch? Is it because it had already been programmed? How do you do a full istall and overwrite everything?

toblum, 2020-03-31:
Hi @grumpazoid,

the pin setting in code is mainly for legacy reasons. For newer versions you can do the config via REST call as described here: https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

Greetings

HiranChaudhuri, 2020-04-01:
> Question is why did it not take the correct pin from the sketch? Is it because it had already been programmed? How do you do a full istall and overwrite everything?

The ESP has several memory regions. There is at least one for firmware code and some for user data - this is where configuration is stored. It is not McLighting specific. If you are unaware about this you may not know the difference between flashing and flashing after a full reset. So there is a chance McLighting just used settings that were left from other firmware and got to wrong conclusions.
How exactly the flash/full reset get triggered your side depends on the tools you are using.

grumpazoid, 2020-04-02:
@toblum @HiranChaudhuri Thank you - all makes sense now.