toblum/McLighting #421: AP & off line operation

iansexton, 2019-06-26:
Hi
I recently noticed that mclighting doesn't work offline (ie without internet connection)
Looking at the code I see that the html calls web resources. Is this what stops it working? I know I could work it out myself but I'm feeling lazy...
Presumably this is why it doesn't work in AP mode?
Wouldn't this be simple to fix?
Cheers
Ian

toblum, 2019-06-29:
Hi @iansexton,

the standard UI needs network connection. Maybe I'll change that in the future.
You can use yout hotspot from a mobile phone to make it work everywhere.

There were also other attempts to create a UI without network connection required: https://github.com/toblum/McLighting/issues/237

There also a simple android app: https://github.com/toblum/McLighting/tree/master/clients/Android

Regards
Tobias

closed 2019-06-29 by toblum

iansexton, 2019-07-02:
@toblum/McLighting <McLighting@noreply.github.com>
 Thanks for this. How did you generate the UI?
There are lots of places in the UK where a mobile phone has no access :(
Couldn't the remote resources be bundled with mclighting? I notice that one
is a very small file.
I'll take a look at #237 <https://github.com/toblum/McLighting/issues/237>

Cheers
Ian

toblum, 2019-07-02:
Hi @iansexton,

I'll have a look on a versio without dependencies in the future. This alternative UI should also have no dependencies: https://github.com/toblum/McLightingUI but it's bigger in overall size.

Regards
Tobias