toblum/McLighting #423: Home Assistant and Google Home Assist Help

deppen12, 2019-07-03:
as i'm using McLighting in my home . now i want to use Home assistant and google Assistant . 
now problem:-
               1. my esp boards IP assign dynamically which is " 192.168.1.232" can't use static IP. and as per HA script we need static IP

               2.  For MQTT , i'm using http://cloudmqtt.com. i can control my board from cloud site . now i want to use google assist which i'm not able to do. IFTTT.com don't have any thing for cloudmqtt or i'm not able to find out

toblum, 2019-07-05:
HI @deppen12,

I'm not sure what you're trying to achieve. And I think this isn't really a McLighting issue and more related to HA.
Sorry, but think I can't help here.

Regards
Tobias

closed 2019-07-05 by toblum

deppen12, 2019-07-07:
you can't help me but other can