toblum/McLighting #424: PROGMEM edit is broken

sehraf, 2019-07-04:
There are two entries that add an action for `/edit`:
https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/McLighting.ino#L588-L590
vs
https://github.com/toblum/McLighting/blob/090bddd32b2705f5edb7ec999cb26c50dc470151/Arduino/McLighting/McLighting.ino#L682-L690
The first one is deprecated as it cannot handle PROGMEM edit.htm and should be removed.
Currently the first entry is used and this resulting in 404.

toblum, 2019-07-05:
Hi @sehraf,

thank you for reporting. I'll add that in the next version 2.2.5.

Regards
Tobias

closed 2019-07-05 by toblum

toblum, 2019-07-05:
@sehraf I just published the new release: https://github.com/toblum/McLighting/releases/tag/v2.2.5