toblum/McLighting #426: LEDs nicht steuerbar!

Duesendaniel, 2019-07-05:
McLighting auf D1 mini  hochgeladen (PIN 4, NUMLEDS 30), Wifi konfiguriert, soweit so gut. 
LED Stripe angeschlossen, nur die erste LED leuchtet, Steuerung nicht möglich. 

LED Stripe funktioniert mit anderem D1 mini und FastLED .

Duesendaniel, 2019-07-05:
Habe zum Test die Version 2.1.5 mit älteren Libaries auf den D1 Mini hochgeladen (PIN D4, NUMLEDS 30) und es funktioniert. Die aktuelle Version bekomme ich nicht in den Griff.

toblum, 2019-07-05:
Hallo @Duesendaniel,

es gab mit Version 2.2.1 Änderungen im Setup. Das hat leider noch im Wiki gefehlt. Die Anzahl der LEDs und den PIN kann man jetzt dynamisch während der Laufzeit ändern. 
Weitere Informationen findest du hier:
https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

Viele Grüße
Tobias

Hello @Duesendaniel,

there were changes in the setup with version 2.2.1. Unfortunately this was missing in the Wiki. The number of LEDs and the PIN can now be changed dynamically during runtime. 
Further information can be found here:
https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

Many greetings
Tobias

closed 2019-07-05 by toblum

Duesendaniel, 2019-07-06:
Hallo Tobias,

danke für die schnelle Info, es funtioniert jetzt. Tolles Projekt.

Gruß Hans