toblum/McLighting #427: Error compiling sketch...I don't understand why

leonealex, 2019-07-07:
Arduino:1.8.9 (Mac OS X), Scheda:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Higher Bandwidth, Disabled, None, Only Sketch, 115200"

Opzioni di compilazione cambiate, ricompilo tutto
/Users/Alessandro/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: libraries/EEPROM/EEPROM.cpp.o: in function `EEPROMClass::begin(unsigned int)':
/Users/Alessandro/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/libraries/EEPROM/EEPROM.cpp:71: undefined reference to `_FS_end'
collect2: error: ld returned 1 exit status
exit status 1
Errore durante la compilazione per la scheda NodeMCU 1.0 (ESP-12E Module).

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

toblum, 2019-07-07:
Hi @leonealex,

I didn't see this error before. Which version of the ESP core are you using? I tried with the 2.5.2 without problems.

Greetings
Tobias

ahauser31, 2019-08-18:
I have the same issue... I know where it comes from, but not why I have it.
The esp8266/Arduino changed their linker scripts and renamed _SPIFFS_end to _FS_end in core 2.5.2 as far as I know. When I compile with PlatformIO, I get in the terminal that Package "framework-arduinoespressif8266 2.20502.0 (2.5.2)" is used. Then I get the linker error above.
When I go to my plafromIO / packages folder and to the framework-arduinoespressif8266 folder, I can find the linker scripts under "tools/sdk/ld" - and there, the scripts define _SPIFFS_end, which is the old version. This makes no sense and I don't know where this comes from. I tried re-installing the framework, but no luck. Is anyone else having that issue as well apart from @leonealex and me?