toblum/McLighting #428: Changing color does not work when setting brightness at the same time.

M4lik, 2019-07-09:
I've connected my McLighting desk lamp with my Home Assistant via MQTT. Furthermore my HA is connected to Google Assistant via Nabu Casa.

If i now give GA the command "set my desk lamp to green", following MQTT message is recorded at my broker:

>home/Schreibtischlampe_ha/state/in
```json
{
  "state" : "ON",
  "color" : {
    "r" : 0,
    "g" : 255,
    "b" : 0
  },
  "brightness" : 255
}
``` 
Which is in return answerd by

> home/Schreibtischlampe_ha/state/out

```json
{
  "state" : "ON",
  "color" : {
    "r" : 0,
    "g" : 255,
    "b" : 0
  },
  "brightness" : 255,
  "color_temp" : 333,
  "speed" : 196,
  "effect" : "Static"
}
```

But the color does NOT acutally change.
I think this is connected to setting the brightness, because if i manually send

>home/Schreibtischlampe_ha/state/in
```json
{
  "state" : "ON",
  "color" : {
    "r" : 0,
    "g" : 255,
    "b" : 0
  }
}
``` 
I get the same response, but the color of the strip changes.

debsahu, 2019-07-10:
Ahh ok HA has now changed the way it sends brightness. Earlier they were sent separately.

M4lik, 2019-07-10:
I think this is not a problem with HA but with Google Asisstant in particular as it sends both, color and brightness, in on command to HA wich only forwards it to McLighting/MQTT.
Nontheless is this also an issue with McLighting as it seems it can't process color and brightness at the same time.