toblum/McLighting #429: The system restarts automatically "static red"

donotoverdriveit, 2019-07-11:
The problem is already described here, but I will do it more closely.

When the LEDs are off, they sometimes start automatically and turn red statically. I think that happens, too, if you unplug it from the plug and plug it in again. The LEDs will then glow statically red.

I have it permanently on power and could find no damage to cables.

Even if I set it off via the web interface, it happens that the LEDs sometimes turn on automatically (static red).

Sometimes everything goes well one day, sometimes the error occurs after half an hour.

Cypris2010, 2019-08-11:
I have the same issue. When I unplug / plug the leds got to static red.

Cypris2010, 2019-08-11:
`#define ENABLE_STATE_SAVE 1  `
This solved it for my setup.

donotoverdriveit, 2019-10-20:
Thank you I will test it!

donotoverdriveit, 2019-10-20:
Thank You! Can you show me more exactly where I can find and change this?

donotoverdriveit, 2019-10-20:
> 
> 
> `#define ENABLE_STATE_SAVE 1 `
> This solved it for my setup.

ryancasler, 2019-10-23:
If you are using a Wemos D1 mini, make sure that you have it set for 4M(1M spiffs).  No Spiffs is the default for some reason now.  With that setting, no settings will save in between power cycles.

donotoverdriveit, 2019-10-24:
Hmm this option was already set for me:
[image]

donotoverdriveit, 2019-10-27:
I now have the problem that the set LED number is no longer correct after I set to 1 MB SPIFFS. Do you have any idea why this is possible?

toblum, 2019-10-27:
Have you set the pixel number via pixelconf as it's described here? https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

donotoverdriveit, 2019-11-07:
Yes. Now I have set to 2 MB SPIFFS. Now it works!

donotoverdriveit, 2020-01-11:
Hello again. With "2 MB SPIFFS" it always saves the last state, even if it is disconnected from power. Works well. The number of LEDs is also taken over correctly (that is not done with 1 MB SPIFFS). Unfortunately, I noticed that the Wemos D1 mini crashes after a while and continues the previous state. It's not that bad, but I don't like it. Is there a solution for this?

ryancasler, 2020-01-16:
Because it is not actually saving your settings when you set it to 2M spiffs.  There's only 4M total. The program is quite large and takes up a lot of the space.  You shouldn't need more than 1M spiffs to save all the data that McLighting needs.  Are you finding that the 1M spiffs setting is not working?

donotoverdriveit, 2020-01-16:
Hello, thanks for the reply.  If I set it to 1M, the set number of LEDs are not correct.  Then fewer LEDs are used than set.

ryancasler, 2020-01-16:
> Then fewer LEDs are used than set.

I'm sorry, I'm not understanding.  How many LEDs is it using?  The default?

donotoverdriveit, 2020-01-19:
Exactly, I've set more LEDs. But with 2MB it always uses the standard number (I think 14 or 20 LEDs?). Ans with 1MB it always starts with the standard (static red) after some time. With 3MB everything was great, but as described, after some time, short crash and restart.

donotoverdriveit, 2020-07-15:
Exactly, I've set more LEDs. But with 2MB it always uses the standard number (I think 14 or 20 LEDs?). Ans with 1MB it always starts with the standard (static red) after some time. With 3MB everything was great, but as described, after some time, short crash and restart.

**No solution?**