toblum/McLighting #439: error compiling for board NIodeMCU 1.0 (ESP-12E Module)

AnandaThalia54, 2019-08-06:
Arduino: 1.8.9 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/anandathalia/Downloads/arduino-1.8.9/arduino-builder -dump-prefs -logger=machine -hardware /home/anandathalia/Downloads/arduino-1.8.9/hardware -hardware /home/anandathalia/.arduino15/packages -tools /home/anandathalia/Downloads/arduino-1.8.9/tools-builder -tools /home/anandathalia/Downloads/arduino-1.8.9/hardware/tools/avr -tools /home/anandathalia/.arduino15/packages -built-in-libraries /home/anandathalia/Downloads/arduino-1.8.9/libraries -libraries /home/anandathalia/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path /tmp/arduino_build_231133 -warnings=all -build-cache /tmp/arduino_cache_934921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=/home/anandathalia/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=/home/anandathalia/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=/home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=/home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1 -verbose /home/anandathalia/Arduino/sketch_aug05a/sketch_aug05a.ino
/home/anandathalia/Downloads/arduino-1.8.9/arduino-builder -compile -logger=machine -hardware /home/anandathalia/Downloads/arduino-1.8.9/hardware -hardware /home/anandathalia/.arduino15/packages -tools /home/anandathalia/Downloads/arduino-1.8.9/tools-builder -tools /home/anandathalia/Downloads/arduino-1.8.9/hardware/tools/avr -tools /home/anandathalia/.arduino15/packages -built-in-libraries /home/anandathalia/Downloads/arduino-1.8.9/libraries -libraries /home/anandathalia/Arduino/libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -ide-version=10809 -build-path /tmp/arduino_build_231133 -warnings=all -build-cache /tmp/arduino_cache_934921 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs.path=/home/anandathalia/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=/home/anandathalia/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=/home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=/home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1 -verbose /home/anandathalia/Arduino/sketch_aug05a/sketch_aug05a.ino
Using board 'nodemcuv2' from platform in folder: /home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2
Using core 'esp8266' from platform in folder: /home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2
Build options changed, rebuilding all
Detecting libraries used...
/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_231133/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/nodemcu /tmp/arduino_build_231133/sketch/sketch_aug05a.ino.cpp -o /dev/null
Generating function prototypes...
/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_231133/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/nodemcu /tmp/arduino_build_231133/sketch/sketch_aug05a.ino.cpp -o /tmp/arduino_build_231133/preproc/ctags_target_for_gcc_minus_e.cpp
/home/anandathalia/Downloads/arduino-1.8.9/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_231133/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/anandathalia/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip2/include -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_231133/core -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -fno-exceptions -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10809 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/cores/esp8266 -I/home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/variants/nodemcu /tmp/arduino_build_231133/sketch/sketch_aug05a.ino.cpp -o /tmp/arduino_build_231133/sketch/sketch_aug05a.ino.cpp.o
Compiling libraries...
Compiling core...
/home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python /home/anandathalia/.arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/signing.py --mode header --publickey /home/anandathalia/Arduino/sketch_aug05a/public.key --out /tmp/arduino_build_231133/core/Updater_Signing.h
fork/exec /home/anandathalia/.arduino15/packages/esp8266/tools/python/3.7.2-post1/python: no such file or directory
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

AnandaThalia54, 2019-08-06:
can anyone help me to fix it?

CristeaAlex7, 2019-09-14:
I also had problems with compiling, but I tried to flash with https://youtu.be/7axaXO8cFo8 and it helped