toblum/McLighting #443: How to make a static IP?

vasya198424, 2019-08-27:
How to make a static IP? In order not to look for it in the settings of the router or in the port monitor

Teneh, 2019-09-02:
In the FAQ: 

How do I set a static IP for my ESP?
Just uncomment //#define WIFIMGR_SET_MANUAL_IP in definitions.h (20-ish Line) and edit the block below that.

#ifdef WIFIMGR_SET_MANUAL_IP
  uint8_t _ip[4] = {192,168,0,128};
  uint8_t _gw[4] = {192,168,0,1};
  uint8_t _sn[4] = {255,255,255,0};
#endif

vasya198424, 2019-09-02:
> In the FAQ:
> 
> How do I set a static IP for my ESP?
> Just uncomment //#define WIFIMGR_SET_MANUAL_IP in definitions.h (20-ish Line) and edit the block below that.
> 
> #ifdef WIFIMGR_SET_MANUAL_IP
> uint8_t _ip[4] = {192,168,0,128};
> uint8_t _gw[4] = {192,168,0,1};
> uint8_t _sn[4] = {255,255,255,0};
> #endif

Thanks, but
I did not write correctly. How to set a static address for a web interface?