toblum/McLighting #446: How many led Can drive with E1.31

yusufsulaiman, 2019-09-05:
Hi. I try to use E1.31 and maximum led can drive is 400 pixels. Can much more drive led?

AndrewVlz, 2019-10-03:
> Hi. I try to use E1.31 and maximum led can drive is 400 pixels. Can much more drive led?

Sorry, I don't know answer for your question.

But...

Some months ago you asked:

https://github.com/vvip-68/GyverMatrixWiFi/issues/5

I met the same problem. I asked vvip-68 about it.

He gave me advice to install latest version of Arduino IDE and to update ESP8266 library to v. 2.5.2 and - no more errors.

You can find this here (in Russian, sorry):

https://github.com/vvip-68/GyverMatrixWiFi/issues/9

mherweg, 2019-12-12:
try to increase END_UNIVERSE  in definitions.h

(170 LEDs per DMX Universe)