toblum/McLighting #451: Led strip flickering random

nikitawetzel, 2019-10-16:
Hello i have a issue.

First try on installing I used 12 led pixels it worked then I go up to mz big one with 83 led pixels first time work then I trying to update code to the 83 num_led after this led strip didn't turn on at all than I thought  my wiring was bad I clean all up and still  didn't work then I see in serial monitor that the led pin changed by it self then I put the new one in them strip was turning on but flickering and spamming text ind serial monitor constantly changing rgb color I've tried to change the effect didn't work then I put out the button then it work but bugging around rainbow cycle didn't move only by one pixel and go back i don't know how to fix it if try to finde the value that changed by it self but can't finde it if tried to upload new code but it seems to don't affect the sketch anything saved in a online file or something I hope someone have same issue  and can help me thank u

jefferdo, 2020-02-14:
Got the same issue

akomylator, 2020-03-13:
Та же проблема,300 led мерцают случайным образом

OGgerald, 2020-09-25:
Mhhhh, same Problem here - Led´s are just flickering. Gui seems to be fine and working but the output does strange things.
Is there meanwhile a solution to this Problem?
Thanks in advance Gerald