toblum/McLighting #454: Lights turned off by default after power failure?

guardiande, 2019-10-27:
Hi,

Is it possible to have to lights turned off by default when the device recovers after a power loss?

toblum, 2019-10-27:
Hi @guardiande,

the strip is initialied here: https://github.com/toblum/McLighting/blob/eaa9f814d841ff6336f1789ac6e85775514b50ea/Arduino/McLighting/McLighting.ino#L293
You may just set  the color to 0,0,0 by default. That should work,

guardiande, 2019-10-27:
Hi @toblum,

Thanks for the quick response!

That does not work, since at the end of the setup function the strip configuration gets restored if enabled.

I added a `initStrip()` as the very last statement in the setup function, which did not help. After also adding a `mode = OFF` the strip does not power on after a power failure.

toblum, 2019-10-27:
Yes, I was wrong. The function readStateFS() does the restoration of the saved values when powering up, so you may be able to override the standard behaviour here:
https://github.com/toblum/McLighting/blob/eaa9f814d841ff6336f1789ac6e85775514b50ea/Arduino/McLighting/request_handlers.h#L1432