toblum/McLighting #457: Two small led

alfredolo, 2019-11-05:
hi, I have 16 independent LED strips and they all work. With a button on the main power I turn off everything when not in use. When I turn on some strips have the first led lit up at maximum brightness in green. Others no. This happens only during the boot of the wemos I use. When the program starts everything is perfect.
Another problem that I encountered during installation is with some LED strips that when turned on are completely white at maximum brightness and without control. You must change the strips with others that are exactly the same to make them work. When I try the LED strips that only light up white, they only work with one logical bidirectional level shifter, while with another perfectly equal and 74hct14 dont.
Do you have any idea why?

I am using :
- wemos d1 mini pro v2
- power shield for 12v power
- 74hct14 as level shifter
- 1000uf connected to power suply 12v
- 470 ohm resistor close to led strips
- ws2815b 12v

Thank you

Cam1011, 2019-11-10:
have you calculated the total amount of current all leds will draw at maximum brightness, it sounds possible that you have the correct voltage but not enough current to handle all the leds. i recommend disconnecting a few strips and seeing if the problems go away. this will tell you if it is drawing to much current without doing all the math.

alfredolo, 2019-11-11:
Thanks for reply.
Yes i calculated it. One of the white strips is 73 leds total with a 35w 12v power suply.
Another one is 48 leds total with 60w 12v power suply.
The strange think if i change firmware with wled both works fine.

ryancasler, 2019-11-11:
Are you shifting the level of the control line from 5v to 12v or from 3.3v to 12v?  Because the esp8266 uses 3.3v logic so you would have to change from 3.3v to 12v.

alfredolo, 2019-11-12:
3.3 to 5v.
ws2815b use 12v for power and 5v for signal line. This is what people says on the web

ryancasler, 2019-11-13:
In that case, try removing the level shifter.  I have never needed one for 5v strips and when I did try to use it I had a bunch of problems since it wasn't operating fast enough for the messages from McLighting.  5v ws28212b run fine with 3.3v on the control line.  You shouldn't damage anything by using too low a voltage.  If anything, it just won't work.  But it's worth a shot.

alfredolo, 2019-11-13:
I tryed it. Some strips works at 3.3v, some don't. I have changed 74htc14 with 74ahct125n , this is fastest than leds. Always the same problem. Some strips only light up white at full bright.
Some strips are perfect.