toblum/McLighting #459: cannot compile

cjs1502, 2019-11-16:
In file included from C:\Users\purvr\OneDrive\Documents\Arduino\libraries\arduinoWebSockets-master\src\libb64\cencode.c:9:0:

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:88:1: error: unknown type name 'uint32_t'

 inline uint32_t esp_get_cycle_count() __attribute__((always_inline));

 ^

In file included from C:\Users\purvr\OneDrive\Documents\Arduino\libraries\arduinoWebSockets-master\src\libb64\cdecode.c:9:0:

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:88:1: error: unknown type name 'uint32_t'

 inline uint32_t esp_get_cycle_count() __attribute__((always_inline));

 ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:89:1: error: unknown type name 'uint32_t'

 inline uint32_t esp_get_cycle_count() {

 ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h: In function 'esp_get_cycle_count':

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:90:3: error: unknown type name 'uint32_t'

   uint32_t ccount;

   ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:89:1: error: unknown type name 'uint32_t'

 inline uint32_t esp_get_cycle_count() {

 ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h: In function 'esp_get_cycle_count':

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h: At top level:

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:112:24: error: unknown type name 'uint32_t'

 void precache(void *f, uint32_t bytes);

                        ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:90:3: error: unknown type name 'uint32_t'

   uint32_t ccount;

   ^

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h: At top level:

C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\cores\esp8266/core_esp8266_features.h:112:24: error: unknown type name 'uint32_t'

 void precache(void *f, uint32_t bytes);

                        ^

Multiple libraries were found for "ESP8266WebServer.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WebServer
Multiple libraries were found for "EEPROM.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\EEPROM
Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266mDNS
Multiple libraries were found for "WebSockets.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\arduinoWebSockets-master
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\ArduinoJson
Multiple libraries were found for "ESPAsyncUDP.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\ESPAsyncUDP-master
Multiple libraries were found for "ESPAsyncE131.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\ESPAsyncE131-master
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266WiFi
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\DNSServer
Multiple libraries were found for "WiFiManager.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\WiFiManager-master
Multiple libraries were found for "WS2812FX.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\WS2812FX
Multiple libraries were found for "Ticker.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\Ticker
Multiple libraries were found for "Hash.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\Hash
Multiple libraries were found for "AsyncMqttClient.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\async-mqtt-client-master
Multiple libraries were found for "ESPAsyncTCP.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\ESPAsyncTCP-master
Multiple libraries were found for "ESP8266HTTPUpdateServer.h"
 Used: C:\Users\purvr\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.1\libraries\ESP8266HTTPUpdateServer
Multiple libraries were found for "Adafruit_NeoPixel.h"
 Used: C:\Users\purvr\OneDrive\Documents\Arduino\libraries\Adafruit_NeoPixel
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

ryancasler, 2019-11-16:
What did you have your board's memory set for?  Did you have it set for 4M/no spiffs or did you change it to 4M/1M spiffs?   Which version of the ESP8266 library do you have?  What did you have defined for you LED Pin number?  Or were you trying to use DMA (which doesn't work)?  If so, try just assigning to GPIO3 and then connecting to RX.  That is what I did.

cjs1502, 2019-11-16:
i tried using 4M/1M and 4M/no spiffs, neither of those worked form me.
esp8266 version 2.6.1.
No change in led pin number (gpio14)
i am getting error while compiling, not while uploading.
seems like some version of any library is causing this problem.

cjs1502, 2019-11-16:
seems like new version of esp8266 package doesn't work with mcl.
i downgraded it to 2.5.0 and it worked for me.
anyway thank you @ryancasler .

closed 2019-11-16 by cjs1502