toblum/McLighting #461: Led Reagieren nicht.

Pete28, 2019-12-01:
Hallo Ich habe alles nach deiner Anleitung gemacht, leider lautet immer nur eine Led obwohl ich 4 eingetragen habe.
Auch ist diese immer Weiß und ich kann sie auch nicht steuern. 
Kann mir hier jemand bei dem problem helfen?

toblum, 2019-12-02:
Hallo @Pete28,
reagiert die Oberfläche  und siehst du Ausgaben am Serial Monitor wenn du die Farbe änderst?
Gruß
Tobias

mrameis, 2020-02-09:
Hallo @Pete28, 
Ich hatte das selbe Problem, ich hatte vergessen die beiden GND's miteinander zu verbinden. 
LG Michael

red-family, 2020-02-17:
Hallo,
sicher das du den richtigen Pin benutzt hast ?
Welchen ESP hast du benutzt ?

Floppynator, 2020-03-21:
prüf mal den PIN. Wenn du einen WEMOS nutzt ist es nicht D4 (z.B.) sondern eine Integer Zahl. Am besten bei WEMOS auf der Seite die PINOUTS ansehen.

ridian2020, 2020-05-11:
GIbt es hierzu neue ERkenntnisse. Habe leider das selbe Problem mit meinem Wemos D1 Mini.
Habe es mit D4 sowie dem Integer 2 laut PIN Plan (https://docs.wemos.cc/en/latest/d1/d1_mini.html) versucht. Leider immer noch ohne Erfolg.

mrameis, 2020-05-11:
Probiers einmal wenn du diesen Befehl verwendest zum festlegen des Pin's  http://YOUR_ESP8266_HOSTNAME_OR_IP/pixelconf?ct=16&rgbo=grb&pin=4 (Muss natürlich noch angepasst werden)

ridian2020, 2020-05-11:
Danke @mrameis für deine Antowrt.
Hab den Fehler gefunden.

Hatte die WS2812FX falsch bzw. nicht gesetzt:
von 
`#define USE_WS2812FX_DMA 0`
auf
`#define USE_WS2812FX_DMA 2 `
geändert und schon läufts.

itscrizze, 2021-05-27:
ridian2020. Kannst du mir vielleicht sagen wo du den Befehl gefunden hast, weil ich habe das Problem das meine LEDs auch nur weiß leuchten, im serial Monitor sehe das ich die Farbe änder aber bei den LEDs kommt nix an. Zu Info ich habe eine 5 v sondern 12v LEDs ist das vielleicht das Problem, die LEDs laufen über ein 12v netzteil