toblum/McLighting #465: Stair lightning with McLightning over iobroker

Nussert, 2019-12-14:
Dear Members,

i want to realize stair lighting. 
When you go up or down the stairs, the led stripe should light up step by step. 
After a short time, the stairs go out one by one. 
This function should be started by a PIR.
Can this solved by a script in the iobroker?
Does anyone have any experience for this idea?

Thanks a lot.

ThmsD, 2019-12-14:
That's not any kind of issue and should not be posted here on GitHub. Please refer to forums for ioBroker for example.

Alveran81, 2019-12-14:
I think its not a problem of iobroker. The question is how to make custom effects. Maybee Led 1-10 get lighter slowly then led 11-20 an, 21-30 ....

bpohvoodoo, 2020-02-02:
Maybe you should read the wiki for Websocket_API, which is the same as MQTT_API

Payload: R[r_start][r_end][hexrgb] [...]
Light multiple LED ranges in the given colors.

Where <r_start> is the first index of the range (start index is 00). Where <r_end> is the last index of the range. Where <hexrgb> is the color as a hex value, e.g. 04d2ff.

This command can be daisy chained!

Example: R0009ffffffR1019ff0000 lights the first 10 LEDs white and the next 10 red Result: OK