toblum/McLighting #473: installing mc lighting with hassio

nickpesman, 2020-02-20:
i have been struggling with this for a while now and i can't get it to work is there someone that can maby give me some help or has a guide to setup mc lighting in my home assistant because i just can't get it to work.......

XaeroDennis, 2020-03-11:
Hope you already got it to work? 
maybe there is a chance i can help. just hmu

123vollhorst, 2020-04-03:
Hi there, I am also struggling with the HA integration as HA is not finding any new lights. Maybe some one can help me with this. 

I followed [the McLighting guide](https://github.com/toblum/McLighting/wiki/Homeassistant-integration) and have general access to the lights via WebUI. I changed the Hostname to my one and the lights are connected to the MQTT server. As I integrated different lights via mqtt auto-discovery before, this feature should also work. 

BUT: As the topics are set to "home/McLightingRGBW_1_ha/state/out" (e.g., in my case) and auto-discovery prefix is set to "homeassistant", how can then HA find the light? For example, my Milights are integrated via "homeassistan/light/milight_hub/...." why isnt there a topic like "homeassistant/light/McLighting....."? Do I miss something here?
Thanks
KeLvin