toblum/McLighting #474: which Websocket library

ramsbod, 2020-03-07:
You say the websocket library to use is version 2.1.4 by links2004.  The newest version of this library is 0.4.15.   The only wesbsocket library I can find with a version 2.1.4 is by Markus Sattler, just called Websockets, not arduinoWebsockets.   Which is the correct one to use.  The reason I am asking is sometime when I change the mode of the LEDs the ESP looses wifi connection.

toblum, 2020-03-07:
Hi @ramsbod,

have a look here: 2.1.4 is the latest version: https://github.com/Links2004/arduinoWebSockets/releases/tag/2.1.4

I'm not sure if your issues are related to the websockets lib version, but give it a try. Often these things happen due to voltage problems.

Best regards
Tobias

ramsbod, 2020-03-07:
I am using a 2 Amp 5V supply to power the level shifter and a 1Amp 5V to 3.3V buck converter to power an esp8266 and a 6Amp 12V supply to power 72 pixels.  The Ground connections are all commoned together. Not sure why power would be a problem.