toblum/McLighting #478: Enable TV over MQTT

gulliverrr, 2020-03-22:
I have everything working but for a strange reason I cannot enable the TV mode over MQTT.
If I send:

    {"state":"ON","color":{"r":255,"g":86,"b":39},"brightness":44,"color_temp":327,"speed":162,"effect":"Chase Random"}
it works, but if I send to the same topic:

    {"state":"ON","color":{"r":255,"g":86,"b":39},"brightness":44,"color_temp":327,"speed":162,"effect":"TV"}

nothing changes. I tried gradually omitting color, brightness, color_temp and speed from the message. No luck.
Serial log spits this:

    MQTT: Recieved [home/LivingRoomLEDs/state/in]:     {"state":"ON","color":{"r":255,"g":86,"b":39},"brightness":44,"color_temp":327,"speed":162,"effect":"TV"}
    Saving cfg: {"mode":1,"strip_mode":0,"brightness":44,"speed":162,"red":255,"green":177,"blue":109} Success!
    MQTT: Send [home/LivingRoomLEDs/state/out]: {"state":"ON","color":{"r":255,"g":177,"b":109},"brightness":44,"color_temp":327,"speed":162,"effect":"Static"}
    Heap size: 17568

Not sure about this `"strip_mode":0`
All other effects work fine. Web interface works for TV so I assume it is compiled/included correctly. Any advice?

toblum, 2020-03-31:
Hi @gulliverrr,

the TV mode is a bit special as it's no native WS2812FX effect.
I'm not fimiliar with the HA MQTT, bu you should be able to use =tv command as described here: https://github.com/toblum/McLighting/wiki/WebSocket-API#payload-lightmode

Greetings

gitpower2017, 2020-11-30:
Hello,

the "homeassistant" implementation shows you real names.

With me: 

`{"name":"McLWZimmer","schema":"json","state_topic":"home/McLWZimmer_ha/state/out","command_topic":"home/McLWZimmer_ha/state/in","brightness":"true","rgb":"true","white_value":"true","optimistic":"false","color_temp":"true","effect":"true","effect_list":["OFF","Static","Blink","Breath","Color Wipe","Color Wipe Inverse","Color Wipe Reverse","Color Wipe Reverse Inverse","Color Wipe Random","Random Color","Single Dynamic","Multi Dynamic","Rainbow","Rainbow Cycle","Scan","Dual Scan","Fade","Theater Chase","Theater Chase Rainbow","Running Lights","Twinkle","Twinkle Random","Twinkle Fade","Twinkle Fade Random","Sparkle","Flash Sparkle","Hyper Sparkle","Strobe","Strobe Rainbow","Multi Strobe","Blink Rainbow","Chase White","Chase Color","Chase Random","Chase Rainbow","Chase Flash","Chase Flash Random","Chase Rainbow White","Chase Blackout","Chase Blackout Rainbow","Color Sweep Random","Running Color","Running Red Blue","Running Random","Larson Scanner","Comet","Fireworks","Fireworks Random","Merry Christmas","Fire Flicker","Fire Flicker (soft)","Fire Flicker (intense)","Circus Combustus","Halloween","Bicolor Chase","Tricolor Chase","ICU","Autoplay","Custom WS","TV","E1.31","Fire 2012","Gradient","Custom 6","Custom 7","Custom 8","Segment OFF"]}`