toblum/McLighting #479: control over the network with Artnet or tpm2.net?/übers Netzwerk steuern mit Artnet oder tpm2.net?

KoggeHD, 2020-04-10:
GER:
Hallo, bin heute auf das geile Projekt gestoßen und will jetzt mir ein Adressierbaren LED Strip verbauen den ich über mein Netzwerk steuern will. 

Ich muss sagen ich bin in den Bereich echt kein Fachmann, aber gibt es da eine Möglichkeit?

Ich habe etwas nachgeforscht und öfters gelesen das es mit Artnet am besten funktionieren soll, allerdings waren die projekte immer nie gut beschrieben wie man es macht also ehr nicht hilfreich.
Als alternative hatte ich immer von tpm2.net gelesen ob das jetzt besser oder schlechter ist weiß ich halt nicht.

meine Frage ist gibt es eine Möglichkeit Artnet oder tpm2.net in das Projekt MCLighting einzufügen und einzurichten und kann mir jemand sagen wie. 

Ich hoffe jemand kann mir helfen.
MfG Kogge

ENG:
Hello, I came across the horny project today and now want to install an addressable LED strip that I want to control via my network.

I have to say I'm really not an expert in the field, but is there a possibility?

I have researched and read a lot that it should be heard best with Artnet. however, the projects were never well described how you do it.
As an alternative, I had always read about tpm2.net, whether that is better or worse, I don't know.

my question is is there a way to insert and set up Artnet or tpm2.net in the MCLighting project and can someone tell me how.

I hope someone can help me.
Sincerely, Kogge

toblum, 2020-04-16:
Hi @KoggeHD,

thre is a reaaally old branch that supports Artnet, but maybe you want to look into E131 protocol that McLighting supports that is quite similar to Artnet. You can enable the E131 mode via UI.
Have a look at this video from Debashish for more information: https://www.youtube.com/watch?v=lZ09GlO2_8s

Regards
Tobias

KoggeHD, 2020-04-16:
ok thanks.
I'll take a look at it

closed 2020-04-16 by KoggeHD