toblum/McLighting #481: Blink when I change number of led in live

pbenaim, 2020-05-06:
We can change the number of LEDs directly (in live) with http://192.168.0.xxx/pixelconf?ct=NBLED cool.

But to do this gently, for example 10 to 144, I put some intermediates,
But each command turns off/on LEDs, so it's not really absolutely smooth :-)

Maybe one custom animation, in STATIC, but how define the number of leds On?

Thanks for your goods ideas ;-)

toblum, 2020-05-09:
Hi @pbenaim,

I'm not sure what's the idea behind changing the number of LEDs in that way, but I don't think the WS2812FX library that does the animation job for McLighting in the background wasn't built to be used that way.

Maybe you want to have a look at the Range commands in the MQTT and Websocket protocol: https://github.com/toblum/McLighting/wiki/WebSocket-API#payload-numledhexrgbnumledhexrgbnumledhexrgb-

Best regards
Tobias

pbenaim, 2020-05-10:
Work very well 👍 
And with beta I can play with more than 99 🥇

closed 2020-05-10 by pbenaim