toblum/McLighting #483: config file Wifi manager

RubiRubsn, 2020-06-03:
Hi, after uploading the code on my Esp, there is a problem with loading the config file for the Wifi manager. The ESP restart after starting the Config mode. what should i do? I tried older versions of Wifi manager, but the Problem still exists. I tried other ESP's and every esp has the same problem.
-Ruben

toblum, 2020-06-05:
Hi @RubiRubsn,
what exact error are you experiencing? Do you see any error message in the serial monitor?
Did you set the partition scheme in the IDE correct that there is reserved space for the SPIFFS filesystem?

RubiRubsn, 2020-06-05:
Hi, at first i flashed with arduino ide and i thought their was problem with the filesystem but than I used the Platform IO version, so the partition scheme is set by the platformio.ini or? 

thats my serial monitor output:

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
FS File: /neoconfig.json, size: 42B
FS Usage: 502/957314 bytes

Reading neoconfig file... Opened!
neoconfig.json: Parsed!
{"pixel_count":24,"rgb_order":82,"pin":15}*WM: Adding parameter
*WM: strip_size
*WM: Adding parameter
*WM: led_pin
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 6
*WM: Using last saved values, should be faster
Saving Strip cfg: {"pixel_count":24,"rgb_order":82,"pin":15}
*WM: Connection result: 
*WM: 1
Entered config mode
192.168.4.1
McLighting01

Exception (28):
epc1=0x40211196 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000007 depc=0x00000000

ctx: cont 
sp: 3ffffbc0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffd60:  00000009 4021d6ef 00000000 3fff156f  
3ffffd70:  00190001 00000000 00000000 3ffffe00  
3ffffd80:  3fff0be4 0000000a 00000003 3ffe8e34  
3ffffd90:  40104d38 002b6e32 3fff27e4 00000000  
3ffffda0:  3ffefed0 3fff27e4 3fff0be4 4021c7a8  
3ffffdb0:  00000001 002a7a0e 40219b14 3fff0554  
3ffffdc0:  00000001 000000c8 00000000 401006dc  
3ffffdd0:  40207c78 3ffffeb0 3fff0be4 3ffe8e34  
3ffffde0:  00000000 3ffffeb0 00000001 40211238  
3ffffdf0:  00000000 3ffffeb0 3fff0be4 402089bb  
3ffffe00:  00000000 00000000 00000000 3ffe8e34  
3ffffe10:  00000000 3fff0b55 3ffffeb0 4021b68e  
3ffffe20:  3fff0de8 0000042d 0000042d 4010020c  
3ffffe30:  00000001 3ffe8e34 3ffffe7c 401006dc  
3ffffe40:  00000000 3ffe8e34 3ffffe70 4021d6ac  
3ffffe50:  3ffe8e9e 402853d4 3ffffeb0 00000000  
3ffffe60:  00000001 3ffe8e34 3ffffeb0 4021b85d  
3ffffe70:  00000000 00000000 00000000 00000000  
3ffffe80:  00000000 00000000 3fffff50 3fff0be4  
3ffffe90:  4020f5a4 00000000 3ffffeb0 3fff0528  
3ffffea0:  3fff0be4 3fffff50 3ffe853c 4020e63d  
3ffffeb0:  00000000 00000000 3ffe8e34 00000000  
3ffffec0:  3fff18dc 0000000f 00000000 3fff18f4  
3ffffed0:  0000000f 00000000 00000000 00000000  
3ffffee0:  00000000 402213c8 00000000 402213c8  
3ffffef0:  00000000 402213c8 00000000 402213c8  
3fffff00:  00000000 402213c8 00000000 402213c8  
3fffff10:  00000000 00000002 ffffffff fe000001  
3fffff20:  3ffe8e9e 00000000 fe01ef35 4020895c  
3fffff30:  40207ca8 0000000a 3fff3454 3ffe9854  
3fffff40:  3ffe985c 3fff18d4 00000002 3ffe8e9e  
3fffff50:  3ffe983a 3ffe9845 3fff18c4 00000005  
3fffff60:  3ffe8e9e feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  00000000 feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3fff0ca4  
3fffffb0:  3fffdad0 00000000 3fff0c9e 4021e3e4  
3fffffc0:  feefeffe feefeffe 3ffe8794 40100721  
<<<stack<<<

I tried to decode the errors, but im unable to find the .elf file for the MClighting projekt

thanks for your help :)

FostersUK, 2020-06-06:
Seem to be getting the exact same error - device seems to loop restart

toblum, 2020-06-07:
This doesn't look like a problem with the filesystem to me. The configfile was written earlier and is read successfully. Then it goes to WiFiManager config mode where it crashes.
Maybe one library had a recent change that causes these reboots. Maybe you could try the library versions mentioned here: https://github.com/toblum/McLighting#used-libraries as these should work.

tuzhimin, 2020-06-09:
我也遇到同一样的问题，

tuzhimin, 2020-06-09:
void configModeCallback (WiFiManager *myWiFiManager) {
  DBG_OUTPUT_PORT.println("Entered config mode");
  DBG_OUTPUT_PORT.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  DBG_OUTPUT_PORT.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);

//  uint16_t i;
//  for (i = 0; i < strip->numPixels(); i++) {
//    strip->setPixelColor(i, 0, 0, 255);
//  }
//  strip->show();
}
把这个wifi连接里边的for循环注释掉就不会出现错误

RubiRubsn, 2020-06-09:
@tuzhimin Yeah it worked! thanks

closed 2020-06-09 by RubiRubsn

nicoausnrw, 2020-09-15:
Yess, it Works! Thanks @tuzhimin !!

>  把这个wifi连接里边的for循环注释掉就不会出现错误

For all who can not read chinese:
"_Comment out the for loop in this wifi connection and you won't get an error!_"