toblum/McLighting #485: Ich Komme nicht weiter

CapatainArcherx, 2020-06-08:
Hallo erstmal,

Ich bin mit meinem Latain am Ende :-
ich habe einen esp8266MOD 12-F
wenn das aufspeieln fertig ist sehe ich im Monitor das der Sketch läuft
immer und immer wiedere quasi Dauerschleife, eine Verbindung mit Wifi ist nicht möglich um die Einstellungen vorzunehmen.
Wenn ich das Autosrollen im serielen Monitor unterbreche (läuft permanent)
sehe ich folgende Meldung 

fd90:  feefef00 feefeffe 80efeffe feefef00  
3ffffda0:  feefeffe 80efeffe 00000000 00000000  
3ffffdb0:  00000000 00000132 00000000 00000001  
3ffffdc0:  00505345 feefeffe 83efeffe feefef00  
3ffffdd0:  00000000 00000000 00000000 000007d0  
3ffffde0:  00010000 00000000 0050ef00 feefef07  
3ffffdf0:  ffffffff 00000000 00000000 01000001  
3ffffe00:  00010101 01010001 feef0100 3ffe88d3  
3ffffe10:  3ffe88d3 feefef00 feefeffe 80efeffe  
3ffffe20:  feef0001 00007530 feefef00 feefef00  
3ffffe30:  feefeffe 80efeffe 00000100 00000100  
3ffffe40:  feef0100 00000000 00000006 0000000a  
3ffffe50:  3fff3704 feef0301 3fff10ec 4020455c  
3ffffe60:  3ffe8c5e 40230a5c 40230a48 feefeffe  
3ffffe70:  feefeffe 00000000 feefeffe 402013fc  
3ffffe80:  3fff06c8 40230a34 40230a20 feefeffe  
3ffffe90:  feefeffe 00000000 feefeffe feefeffe  
3ffffea0:  feefeffe 00000000 feefeffe feefeffe  
3ffffeb0:  feefeffe 00000000 feefeffe 3ffe9403  
3ffffec0:  3ffe940b 3fff2024 00000002 00000001  
3ffffed0:  3ffe88d3 3ffe93e9 3ffe93f4 3fff36c4  
3ffffee0:  00000005 00000001 3ffe88d3 3ffe93c3  
3ffffef0:  3ffe93be 3fff369c 00000020 00000001  
3fffff00:  3ffe93c8 3ffe93ab 3ffe93a6 3fff3674  
3fffff10:  00000020 00000001 3ffe93b0 3ffe93a1  
3fffff20:  3ffe939c 3fff3664 00000006 00000001  
3fffff30:  3ffe88d3 3ffe915c 3ffe938e 3fff361c  
3fffff40:  00000040 00000001 3ffe88d3 402013fc  
3fffff50:  feefeffe 40230a34 40230a20 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe feefeffe feefeffe  
3fffff80:  feefeffe 3fff06cb feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff1238  
3fffffa0:  3fffdad0 00000000 3fff11f8 40227374  
3fffffb0:  feefeffe feefeffe 3ffe86bc 40101005  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3456, room 16 
tail 0
chksum 0x84
csum 0x84
va5432625
~ld
FS Usage: 0/1953282 bytes

Couldn't find config.json
Coudnt find config.json
WiFiManager config FS Read failure!
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: host
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: strip_size
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: led_pin
*WM: [1] AutoConnect 
*WM: [1] No Credentials are Saved, skipping connect 
*WM: [2] Starting Config Portal 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  FFB_PLEX
*WM: [2] AP has anonymous access! 
*WM: [1] SoftAP Configuration 
*WM: [1] -------------------- 
*WM: [1] ssid:             FFB_PLEX
*WM: [1] password:         
*WM: [1] ssid_len:         8
*WM: [1] channel:          1
*WM: [1] authmode:        
*WM: [1] ssid_hidden:     
*WM: [1] max_connection:   4
*WM: [1] country:          CN
*WM: [1] beacon_interval:  100(ms)
*WM: [1] -------------------- 
*WM: [1] AP IP address: 192.168.4.1
Entered config mode
192.168.4.1
FFB_PLEX

Exception (28):
epc1=0x40221a9e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000007 depc=0x00000000

>>>stack>>>

danach fängt alles von vorne an.
Für hilfe wäre ich Dankbar
MFG

Igor1976-light, 2020-06-08:
Open your .ino with Aduino IDE, go to Tools >> ESP8266 Sketch Data upload....  and i think it work now.

RubiRubsn, 2020-06-09:
hi, Ich habe das selbe Problem, schau mal ein issue tiefer, da wird das schon besprochen. es scheint ein fehler mit der wifimanager lib zu sein

RubiRubsn, 2020-06-09:
Hier :D https://github.com/toblum/McLighting/issues/483