toblum/McLighting #486: Cant Open/Import McLightingApp.aia

huge-11, 2020-06-17:
Hello,
i wanted to check out the McLightingApp on App Inventor but i cant import the aia file. 
Error: The selected project is not a project source file! Project source files are aia files.
And yes i am uploading the aia file. 
Could someone help? 
thanks

toblum, 2020-06-17:
Hi!
Had no problem to import the file. Used the download button on this page https://github.com/toblum/McLighting/blob/master/clients/Android/McLightningApp.aia to get the .aia file.

huge-11, 2020-06-17:
Mmmm yes, that it working. I tried downloading the apk and aia files as a zip and some how that is not working. Neither if i save the target on the link, weird.
But thanks

closed 2020-06-17 by huge-11