toblum/McLighting #489: Changing led pin issue?

huge-11, 2020-09-07:
Hello,
I am having some trouble using mclighting. I want to use pin 14 on an esp03 with an upgraded flash for the strip, but somehow i cannot change the gpio pin on the web page. On the definitio.h i wrote led pin = 14, but on the webpage its always on pin 3. Already made my pcb with pin14? 
Any help please

toblum, 2020-09-07:
Hi @huge-11,

in the recent versions, you can configure the PIN dynamically via HTTP call. 
Have a look here: https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

Greetings

cralex11, 2020-10-17:
I have the same trouble. When I try to access this link :
http://YOUR_ESP8266_HOSTNAME_OR_IP/pixelconf?ct=16&rgbo=grb&pin=4
nothing hopping, and I get this :
File Not Found
URI: /pixelconf

toblum, 2020-10-17:
@cralex11 Did you use the latest version? /pixelconf should always return something.

cralex11, 2020-10-17:
Sorry, Mclightning V2 works perfectly, but V3 from attached link have this problem with port 3, I can't change it from settings and from link.

toblum, 2020-10-17:
Then you should better ask in the https://github.com/FabLab-Luenen/McLighting project for help.