toblum/McLighting #494: McLighting (esp8266)

Mikailo09, 2020-12-06:
Hey, I have problem with McLighting, when I turn on led strip last 15 leds doesn't light.
Led strip is about 10 meters long and my PSU has enough wattage.
Any ideas where can be problem ?
[image]

toblum, 2020-12-06:
Hi @Mikailo09,

did you configure the number of LEDs via /pixelconf?ct=<numberofleds>
Does ist change when you lower the number?

Greetings

Mikailo09, 2020-12-07:
I edited this and everything is still same.
[image]

toblum, 2020-12-07:
Modifying the code is only needed for some special cases. 
You could change NUMLEDS easily vis HTTP call as described here: https://github.com/toblum/McLighting/wiki/Software-installation#configure-pin--rgb-mode--numleds-only-for-v221-or-later

Mikailo09, 2020-12-08:
It works now. Thanks!

closed 2020-12-08 by Mikailo09