toblum/McLighting #500: Build error

cralex11, 2021-02-01:
> Executing task: pio run --environment nodemcuv2 <

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing nodemcuv2 (board: nodemcuv2; framework: arduino; platform: espressif8266@1.8.0)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (1.8.0) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20402.4 (2.4.2) 
 - tool-esptool 1.413.0 (4.13) 
 - toolchain-xtensa 1.40802.0 (4.8.2)
Converting McLighting.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 36 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncMqttClient> 0.8.2
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson> 6.17.2
|-- <WS2812FX> 1.3.2
|   |-- <Adafruit NeoPixel> 1.7.0
|-- <NeoPixelBus> 2.6.0
|   |-- <SPI> 1.0
|-- <WebSockets> 2.3.3
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESPAsyncTCP> 1.2.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Ethernet(esp8266)> 1.0.4
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncE131> 1.0.1
|   |-- <ESPAsyncUDP> 0.0.0-alpha+sha.697c75a025
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncUDP> 0.0.0-alpha+sha.697c75a025
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266HTTPUpdateServer> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WebServer> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <Ticker> 1.0
|-- <WifiManager> 0.14
|   |-- <DNSServer> 1.1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WebServer> 1.0
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio/build/nodemcuv2/src/McLighting.ino.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/WString.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/abi.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/base64.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/cbuf.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/cont.S.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/cont_util.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_eboot_command.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_flash_utils.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_i2s.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_main.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_noniso.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_phy.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_postmortem.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_si2c.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_sigma_delta.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_timer.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_waveform.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring_analog.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring_digital.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pulse.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring_pwm.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/core_esp8266_wiring_shift.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/debug.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/gdb_hooks.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/heap.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/libb64/cdecode.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/libb64/cencode.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/libc_replacements.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/pgmspace.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/sntp-lwip2.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_cache.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_check.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_gc.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs/spiffs_nucleus.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs_api.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/spiffs_hal.cpp.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/time.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/uart.c.o
Compiling .pio/build/nodemcuv2/FrameworkArduino/umm_malloc/umm_malloc.c.o
Archiving .pio/build/nodemcuv2/libFrameworkArduino.a
Indexing .pio/build/nodemcuv2/libFrameworkArduino.a
In file included from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:14:0,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Numbers/convertNumber.hpp:18,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:9,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Variant/SlotFunctions.hpp:8,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Array/ArrayIterator.hpp:7,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Array/ArrayRef.hpp:8,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson.hpp:17,
                 from .pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson.h:9,
                 from /home/cralex/Desktop/McLighting-master/Arduino/McLighting/McLighting.ino:47:
.pio/libdeps/nodemcuv2/ArduinoJson/src/ArduinoJson/Numbers/FloatTraits.hpp:161:5: error: factors causes a section type conflict with __c
     ARDUINOJSON_DEFINE_STATIC_ARRAY(
     ^
In file included from /home/cralex/.platformio/packages/framework-arduinoespressif8266@2.20402.4/cores/esp8266/Arduino.h:256:0,
                 from /tmp/tmp8k0b7pnr:1:
/home/cralex/.platformio/packages/framework-arduinoespressif8266@2.20402.4/cores/esp8266/pgmspace.h:16:51: note: '__c' was declared here
 #define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))
                                                   ^
/home/cralex/Desktop/McLighting-master/Arduino/McLighting/McLighting.ino:681:24: note: in expansion of macro 'PSTR'
     server.send_P(200, PSTR("text/html"), edit_htm_gz, edit_htm_gz_len);
                        ^
*** [.pio/build/nodemcuv2/src/McLighting.ino.cpp.o] Error 1
===================================================================================== [FAILED] Took 3.32 seconds =====================================================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcuv2      FAILED    00:00:03.322
================================================================================ 1 failed, 0 succeeded in 00:00:03.322 ================================================================================
The terminal process "pio 'run', '--environment', 'nodemcuv2'" terminated with exit code: 1.

toblum, 2021-02-20:
Hi @cralex11,

I couldn't reproduce the error, but I saw that the dependencies were updated in the meanwhile. So I updated the platformio.ini to pin the used versions. That worked for me. Maybe you could try again and delete .io directory before.

Greetings
Tobias

cralex11, 2021-02-20:
Thank you, it works!

toblum, 2021-02-20:
Great!

closed 2021-02-20 by toblum