toblum/McLighting #503: connecting Octoprint (raspberry pi 4) to ESP8266 as the controller.

ShadowOfTheDamn, 2021-07-27:
Dear Experts
I want to know if it is possible to connect the an esp8266 with Mclighting to a  raspberry pi 4.
the thing I am looking for is to use this for octoprint ( a 3d printer web interface) somehow as internal connection so I can add a iframe to the octoprint interface and it controls the esp8266. or if it is possible to connect and control the esp8266 via wires to raspberry pi but this connection should be completely done only one time directly to raspberry pi directly without a need for any other hardware except the Raspberry Pi.
[image]

toblum, 2021-07-27:
Hi @ShadowOfTheDamn,

connecting and controlling Octoprint and McLighting wireless is possible via REST/MQTT API. It's not made to be controlled by a wired connection. 

If you can use Wifi for the connection to the ESP then you should definitely have a look at this plugin and WLED as a firmware for the ESP: https://plugins.octoprint.org/plugins/wled/

Greetings

closed 2021-07-27 by toblum

ShadowOfTheDamn, 2021-07-27:
IT DOES NOT DO WHAT i WANT TO, WHY YOU CLOSED THE ISSUE?
I want to control the LED from octoprint GUI with a GUI similar to the McLighting GUI.

toblum, 2021-07-27:
Hi @ShadowOfTheDamn,

as said: this seems not to be the right task for McLighting and more a thing for a Octoprint plugin. I suggested one that works very well. There are definitely many other solutions out there that might fit your needs. 
These issues are for Mclighting related problems. That's why I closed it!

Greetings

ShadowOfTheDamn, 2021-07-27:
Hi again and thanks
you proposed a solution that even does not have a single documentation about how to use it.