toblum/McLighting #504: Change Wifi Network

VeniceNerd, 2021-12-09:
Hi guys,

I've had my McLighting set up for a few years but need to change the WiFi network now. Currently I can still access the McLighting over the old WiFi network so I still have access to the user interface. Is there any way to change the Network name and password somewhere? 

Thanks!

Philipp

toblum, 2021-12-10:
Hi @VeniceNerd,

you may try to call http://Mc_lighting_ip/start_config_ap to enable the configuration AccessPoint like on first setup. Another option would be http://Mc_lighting_ip/reset_wlan to reset credentials that should also bring up the AP on next reboot.
And finally disabling the WiFi network should also bring up the AP.

Greetings

VeniceNerd, 2021-12-10:
> Hi @VeniceNerd,
>
> you may try to call http://Mc_lighting_ip/start_config_ap to enable the configuration AccessPoint like on first setup. Another option would be http://Mc_lighting_ip/reset_wlan to [...]

Hi @toblum thanks so much for the quick reply! I tried "http://Mc_lighting_ip/start_config_ap" and it indeed seems to have enabled the AP as I can connect to it from my iPhone:
[image]
[image]

However, when I try to access 192.168.4.1 nothing will actually load in Safari:
[image]

So while I can connect to the AP I can't seem to connect to the website to change the credentials. Any idea what may be going on here?

Thanks!

Philipp

toblum, 2021-12-13:
Hi @VeniceNerd,

unfortunately this option didnt work well in all versions, the AP worked but the webserver didn't react. Maybe you could try the other option (temporarily disabling the wifi and rebooting McLIghting) to force the AP directly on startup.

Greetings
Tobias